commaai/panda #1343: Compiling with newer versions of GCC results in broken firmware

robbederks, 2023-04-11:
The application firmware image compiled with `arm-none-eabi-gcc` version 10.3 instead of 9.3.1 (default in ubuntu 20.04) does not boot on white pandas (have not tested others). Bootstub seems fine.

pd0wm, 2023-04-13:
Same issue as https://github.com/commaai/panda/pull/1204 ?

olesalscheider, 2023-04-25:
I am seeing the same, but the buffer size is already set to the smaller one.

danielzmod, 2023-04-29:
> The application firmware image compiled with `arm-none-eabi-gcc` version 10.3 instead of 9.3.1 (default in ubuntu 20.04) does not boot on white pandas (have not tested others). Bootstub seems fine.

Tried with grey panda and had similar or same problem. Just seems to be completly dead.

pd0wm, 2023-08-22:
Looks like after https://github.com/commaai/panda/commit/c66b98b2a67441faa4cfcd36c3c9d9f90474cd08 it works again.

robbederks, 2023-08-22:
Still want to look into the root cause at some point to prevent this issue from happening again, so will keep  this open for now.

adeebshihadeh, 2023-08-22:
Once we investigate and fix properly, we’ll also add a latest GCC build to CI.

briskspirit, 2023-08-23:
@pd0wm reported that #1559 fixed the issue. Should we close it or still want to investigate the cause?

adeebshihadeh, 2023-08-23:
No, we definitely still need to look into this.

closed 2026-02-22 by adeebshihadeh