commaai/panda #1441: High temperatures causing the comma 3 to turn on.

rmcc22, 2023-06-03:
When the Comma device is left in a hot car, it will falsely get an ignition signal. The ign pins on the harness are floating.

```
{'logMonoTime': 55684780393,
 'pandaStates': [{'alternativeExperience': 0,
                  'canState0': {'brsEnabled': False,
                                'busOff': False,
                                'busOffCnt': 0,
                                'canDataSpeed': 0,
                                'canSpeed': 500,
                                'canfdEnabled': False,
                                'canfdNonIso': False,
                                'errorPassive': False,
                                'errorWarning': False,
                                'lastDataError': 'noError',
                                'lastDataStoredError': 'noError',
                                'lastError': 'noError',
                                'lastStoredError': 'noError',
                                'receiveErrorCnt': 0,
                                'totalErrorCnt': 0,
                                'totalFwdCnt': 0,
                                'totalRxCnt': 6048,
                                'totalRxLostCnt': 0,
                                'totalTxCnt': 0,
                                'totalTxLostCnt': 0,
                                'transmitErrorCnt': 0},
                  'canState1': {'brsEnabled': False,
                                'busOff': False,
                                'busOffCnt': 0,
                                'canDataSpeed': 0,
                                'canSpeed': 500,
                                'canfdEnabled': False,
                                'canfdNonIso': False,
                                'errorPassive': False,
                                'errorWarning': False,
                                'lastDataError': 'noError',
                                'lastDataStoredError': 'noError',
                                'lastError': 'noError',
                                'lastStoredError': 'noError',
                                'receiveErrorCnt': 0,
                                'totalErrorCnt': 0,
                                'totalFwdCnt': 0,
                                'totalRxCnt': 0,
                                'totalRxLostCnt': 0,
                                'totalTxCnt': 0,
                                'totalTxLostCnt': 0,
                                'transmitErrorCnt': 0},
                  'canState2': {'brsEnabled': False,
                                'busOff': False,
                                'busOffCnt': 0,
                                'canDataSpeed': 0,
                                'canSpeed': 500,
                                'canfdEnabled': False,
                                'canfdNonIso': False,
                                'errorPassive': False,
                                'errorWarning': False,
                                'lastDataError': 'noError',
                                'lastDataStoredError': 'noError',
                                'lastError': 'noError',
                                'lastStoredError': 'noError',
                                'receiveErrorCnt': 0,
                                'totalErrorCnt': 0,
                                'totalFwdCnt': 0,
                                'totalRxCnt': 6049,
                                'totalRxLostCnt': 0,
                                'totalTxCnt': 0,
                                'totalTxLostCnt': 0,
                                'transmitErrorCnt': 0},
                  'controlsAllowed': False,
                  'current': 0,
                  'fanPower': 5,
                  'fanSpeedRpmDEPRECATED': 0,
                  'fanStallCount': 0,
                  'faultStatus': 'none',
                  'faults': [],
                  'gasInterceptorDetected': False,
                  'gmlanSendErrs': 0,
                  'harnessStatus': 'flipped',
                  'hasGpsDEPRECATED': False,
                  'heartbeatLost': False,
                  'ignitionCan': False,
                  'ignitionLine': True,
                  'interruptLoad': 0.03896944597363472,
                  'pandaType': 'dos',
                  'powerSaveEnabled': False,
                  'rxBufferOverflow': 0,
                  'safetyModel': 'noOutput',
                  'safetyParam': 0,
                  'safetyParam2DEPRECATED': 0,
                  'safetyParamDEPRECATED': 0,
                  'safetyRxChecksInvalid': False,
                  'safetyRxInvalid': 0,
                  'safetyTxBlocked': 0,
                  'sbu1Voltage': 0.13600000739097595,
                  'sbu2Voltage': 0.3070000112056732,
                  'spiChecksumErrorCount': 0,
                  'startedSignalDetectedDEPRECATED': False,
                  'txBufferOverflow': 0,
                  'uptime': 7,
                  'usbPowerModeDEPRECATED': 'none',
                  'voltage': 275}],
 'valid': True}
```

```
{'logMonoTime': 57465305028,
 'temperatureSensor': {'sensor': 0,
                       'source': 'lsm6ds3trc',
                       'temperature': 73.05859375,
                       'timestamp': 57464830705,
                       'type': 13,
                       'uncalibratedDEPRECATED': False,
                       'version': 1},
 'valid': True}
```

rmcc22, 2023-06-03:
I am thinking the pin should be pulled up, then read, then pull down, then float.

rmcc22, 2023-06-03:
@robbederks

rmcc22, 2023-06-03:
I will test this.

bool harness_check_ignition(void) {
  bool ret = false;

  // wait until we're not reading the analog voltages anymore
  while (harness.sbu_adc_lock == true) {}

  switch(harness.status){
    case HARNESS_STATUS_NORMAL: {
      uint32_t current_pupdr = current_board->harness_config->GPIO_SBU1->PUPDR;// Save current pull-up/pull-down settings
      set_gpio_pullup(current_board->harness_config->GPIO_SBU1, current_board->harness_config->pin_SBU1, PULL_UP);
      ret = !get_gpio_input(current_board->harness_config->GPIO_SBU1, current_board->harness_config->pin_SBU1);
      set_gpio_pullup(current_board->harness_config->GPIO_SBU1, current_board->harness_config->pin_SBU1, PULL_DOWN); // drain
      current_board->harness_config->GPIO_SBU1->PUPDR = current_pupdr; // Restore previous pull-up/pull-down settings
      break;
    }
    case HARNESS_STATUS_FLIPPED: {
      uint32_t current_pupdr = current_board->harness_config->GPIO_SBU2->PUPDR; // Save current pull-up/pull-down settings
      set_gpio_pullup(current_board->harness_config->GPIO_SBU2, current_board->harness_config->pin_SBU2, PULL_UP);
      ret = !get_gpio_input(current_board->harness_config->GPIO_SBU2, current_board->harness_config->pin_SBU2);
      set_gpio_pullup(current_board->harness_config->GPIO_SBU2, current_board->harness_config->pin_SBU2, PULL_DOWN); // drain
      current_board->harness_config->GPIO_SBU2->PUPDR = current_pupdr; // Restore previous pull-up/pull-down settings
      break;
    }
    default:
      break;
  }
  return ret;
}

robbederks, 2023-06-03:
Not related, this is a known hardware bug in the harness box when the ignition signal is left floating. Please connect it to gnd, which should fix the issue.

closed 2023-06-03 by robbederks

rmcc22, 2023-06-03:
> Not related, this is a known hardware bug in the harness box when the ignition signal is left floating. Please connect it to gnd, which should fix the issue.

I would rather have a sw fix if the hardware permits. Maybe the transistor is in the active region and the internal pull-up would be enough to reduce false ign?

rmcc22, 2023-06-03:
Don't think many people want to cut, crimp and jumper. I will just add a toggle to disable it.

rmcc22, 2023-06-04:
@robbederks you guys already tried setting the internal pull up before reading?

rmcc22, 2023-06-04:
I feel like this a pretty serious unsuspecting issue that needs an immediate fix

vcerchou, 2023-06-04:
Yes, I also encountered this problem

rmcc22, 2023-06-04:
Something like this should prevent it from happening most of the time.

```
bool check_started(void) {
  bool started = current_board->check_ignition() || ignition_can;
  ignition_seen |= started;
  if (ignition_can) {
    harness.use_ignition_line = false;
  }
  return started;
}
```

robbederks, 2023-06-05:
> @robbederks you guys already tried setting the internal pull up before reading?

Yes, it's not related to a floating line, so it won't be solved by setting the pullup. The actual cause is one of the diodes in the harness box which has enough reverse leakage current at high temperatures to reach the threshold voltage of a mosfet. This is completely mitigated by tying the ignition line on the connector side to GND.

We have a proper fix in progress as well, but in the meantime of course we'll be happy to RMA this for people not willing to DIY it. If so, please reach out to support