commaai/panda #1452: CAN-FD: CAN Bus Disconnected during button press send to car

sunnyhaibin, 2023-06-09:
For some HKG CAN-FD, when the resume button press is sent to the car, it would cause "CAN Bus Disconnected" and requires the car to restart to reset the error.

We have attempted to revert this commit, and users are no longer getting this error. We are unsure of the relevance, but so far no user has experienced this issue in many Stop and Go traffic with openpilot sending the resume button to the car.
- https://github.com/commaai/panda/commit/d2c2d5f926537a132a253277998ef350fe866d27

Affected Platforms:
- 2023 Hyundai Palisade
- 2023 Kia Telluride
- 2022 Ioniq 5

Affected Routes:
- `28aa956828c3407d|2023-06-06--09-31-12`
- `abada7a7a4dccf94|2023-06-06--11-15-08`
- `aa5f7d1d9ac7db11|2023-05-30--09-41-25`
- `aa5f7d1d9ac7db11|2023-05-29--12-18-52`

adeebshihadeh, 2023-06-09:
Bus 1 is going into the bus off state. @briskspirit can you look into this?

adeebshihadeh, 2023-06-09:
@sunnyhaibin how did you come up with that commit? did you narrow it down to exactly that one or only tested before/after that one? The commit only changes behavior on init, and doesn't even run on the red panda. It would be helpful if you guys can `git bisect` to the exact commit since it's easy to repro.

sunnyhaibin, 2023-06-10:
> @sunnyhaibin how did you come up with that commit? did you narrow it down to exactly that one or only tested before/after that one? The commit only changes behavior on init, and doesn't even run on [...]

@adeebshihadeh This came up with `git bisect`. It's a bit confusing as the other ones that came up didn't seem to be related with Red Panda, this was the closest one.

adeebshihadeh, 2023-06-20:
I couldn't reproduce this on our EV6 on openpilot master, even sending 50 button msgs every frame. @sunnyhaibin do you have a reliably way to reproduce this? ideally a clean branch based on master

sunnyhaibin, 2023-08-27:
We were attempting to enable openpilot longitudinal control on the Ioniq 6 2023 HDA2 by disabling the ADAS Driving ECU `0x730` via bus 1, the same method used to enable openpilot longitudinal for currently supported HDA2 cars. It seems that as soon as the ECU is disabled (confirmed in cabana and plotjuggler), bus 1 went into the bus off state and there is no traffic on bus 1. This did not affect bus 0 or 2, however.
- https://github.com/commaai/openpilot/pull/29486
- Route ID: `eb4eae1476647463|2023-08-26--20-48-50`

briskspirit, 2023-08-27:
Is it the only ECU on the bus?

sunnyhaibin, 2023-08-28:
> Is it the only ECU on the bus?

It is not, AFAIK. ECUs that broadcast MDPS, SCC, ESP, etc. are also on bus 1.

sunnyhaibin, 2023-08-28:
We have another route that seems to have triggered this issue:
> Hey `@sunnyhaibin` had a canbus error while driving today. I was kind of spamming the off and on instead of letting the car slam on breaks due to the long stock radar. Could have been my actions but wanted to share just in case. I rebooted in the middle of a drive and no issues. Got a few hours of time on most recent update and this is the first fault 

Route ID: `696748e0ac8082fb|2023-08-27--16-38-01`

briskspirit, 2023-08-29:
@sunnyhaibin messaged you with the branch to try

sunnyhaibin, 2023-09-04:
Sent @briskspirit in DM with the routes that have 100% success rates of button sends not putting bus 1 into bus off state, posting them here as well for visibility. https://github.com/commaai/panda/pull/1615 was implemented:
```
696748e0ac8082fb|2023-09-02--19-41-51

696748e0ac8082fb|2023-09-02--20-52-30

696748e0ac8082fb|2023-09-02--22-26-43

696748e0ac8082fb|2023-09-03--12-21-33

696748e0ac8082fb|2023-08-31--14-25-28

28aa956828c3407d|2023-09-03--17-16-47

caf6a54b6d467dbd|2023-09-02--11-59-54
```

sunnyhaibin, 2023-09-05:
A route from sunnypilot 0.9.4.1 with 100% success rates of button sends: `fc19648042eb6896|2023-09-05--14-22-53`

briskspirit, 2023-09-05:
[image]

Data looks promising so far! In this route bus off state happened on first button spam session, CAN core was reset and continued functioning as normal.

briskspirit, 2023-09-06:
merged to master, let me know if something goes wrong

closed 2023-09-06 by briskspirit

VoltIcaRus, 2023-09-07:
@sunnyhaibin @briskspirit  fc19648042eb6896|2023-09-07--16-40-59--3
Something is wrong 
Can you take a look at this? 
I'm not driving and I'm in the passenger seat so I don't remember much, but I think I did a reset without setting the cruise. As a result, the can bus was disconnected and several warning lights came on and off on the dashboard of the vehicle. I immediately restarted the vehicle and it went away.

briskspirit, 2023-09-07:
@VoltIcaRus don't see it to be panda related. by logs panda worked fine, but controls were not allowed. This ticket is the wrong place to post.

VoltIcaRus, 2023-09-07:
Yes If that's not a problem, great
pressed the resume button and the can bus was released and recorded
Glad to hear it's not related, thank you for your research!