commaai/panda #1460: CAN after orientation change

robbederks, 2023-06-14:
Right now if the harness status changes between normal and flipped mode, the CAN buses are mapped wrong.
To fix this, the whole bus_status_t / gmlan / bus flipping stuff should be refactored to support this more easily

adeebshihadeh, 2023-07-02:
Should we just reset on orientation change? I imagine there's a few more things like this that are subtly wrong.

robbederks, 2023-07-02:
sounds a bit too drastic as a final solution, but could be better in the meantime while we don't have this nicely refactored. Should add a log for this though to distinguish it from other kinds of resets

adeebshihadeh, 2024-03-21:
You're right, we should fix this properly.

This adds a pretty meaningful delay to startup https://github.com/commaai/panda/blob/16bd1382dfe13176b172da9fdc749e7f6567c56d/board/drivers/harness.h#L122.

PR #1913 (2024-03-22): Adjust CAN bus orientation live with harness orientation
fixes #1460
closed 2024-03-22 by adeebshihadeh