commaai/panda #1478: SBU1 and SBU2 are not symmetric

robbederks, 2023-06-21:
Looking at the difference between the measured SBU1 and SBU2 voltages from 1000 C3 segments, it's clear that on the dos they are not symmetrical:
[image]

Electrically they should be, so it's likely software-related (pull ups / downs / ...). Doesn't cause issues for the time being, but should be fixed nonetheless.

dzid26, 2024-02-02:
> Electrically they should be, so it's likely software-related (pull ups / downs / ...). 

Or could be due to this..  https://github.com/commaai/panda/issues/1441#issuecomment-1576541065

adeebshihadeh, 2025-07-20:
Is this just on dos? If so, think we can close since it's been stable.

robbederks, 2025-07-23:
will check

robbederks, 2025-07-24:
Doesn't appear to be true for tres, so good to close:

[image]