commaai/panda #1551: Single-Wire CAN issues

pd0wm, 2023-08-04:
I noticed a few issues when trying to communicate with an external device that uses SW-CAN using pin 1 of a white panda.

- Baud rate is 2x too high
- Setting the speed to a different baudrate has no effect, it's stuck on 2x 33.3kbit/s
- The panda does not seem to Ack or receive any messages

Scope screenshot where panda is sending messages. Note baud rate is 2x too high, and ACK from external device. Panda will correctly retry sending the message if an ACK is not received. Yellow = RX, Purple TX, measured behind the `NVC7356` SW-CAN transceiver on the external device.
[image]

Scope screenshot where external device is sending message (in single shot mode to prevent endless re-transmission). No ack, and no messages received from the Python side. Tried sending at both 33.3kbit/s and 66.6kbit/s in case reception was using the correct baud rate.
[image]

Code used to send messages
```python
p = Panda()
p.set_safety_mode(Panda.SAFETY_ALLOUTPUT)
p.can_send(0x0, b"\xaa\xaa", 3)
```

Am I doing something wrong to receive messages? I'm following https://github.com/commaai/panda/blob/master/tests/gmlan_harness_test.py, but I think that uses bit-banged CAN on a regular CAN bus instead of single wire CAN.

adeebshihadeh, 2023-08-05:
Sounds like ti's using the bitbanged GMLAN for some reason: https://github.com/commaai/panda/blob/6e1d35b03331529829e2075d3491f2693034b6e9/board/drivers/gmlan_alt.h#L133.

briskspirit, 2023-08-31:
@pd0wm seems like we can close the issue?

adeebshihadeh, 2023-08-31:
How was this fixed?

pd0wm, 2023-08-31:
Yes, clocks are fixed now. The other issues I mentioned just aren’t supported by but banged gmlan.

I haven’t been able to get that multiplexing working yet, but will make a new issue if necessary.