commaai/panda #1636: Issue with Flashing Panda

ServerDummy, 2023-09-03:
Newer python based panda flashing has a issue with the white/grey panda and will not flash

```
scons: Entering directory `/home/user/panda'
scons: Reading SConscript files ...
scons: done reading SConscript files.
scons: Building targets ...
scons: `board' is up to date.
scons: done building targets.
found 1 panda(s) - ['2e0037000551363338383037']
flashing 2e0037000551363338383037
Traceback (most recent call last):
  File "/home/tkoob/panda/board/./flash.py", line 17, in <module>
    p.flash()
  File "/usr/local/lib/python3.10/dist-packages/pandacan-0.0.10-py3.10.egg/panda/__init__.py", line 513, in flash
    assert os.path.isfile(fn)
AssertionError```

gcormier, 2023-09-06:
Same issue, different error though - white panda

```
greg@grexps:~/panda/board$ ./flash.py 
scons: Entering directory `/home/greg/panda'
scons: Reading SConscript files ...
scons: done reading SConscript files.
scons: Building targets ...
scons: `board' is up to date.
scons: done building targets.
found 1 panda(s) - ['590034000f51373038323837']
flashing 590034000f51373038323837
Traceback (most recent call last):
  File "/home/greg/panda/board/./flash.py", line 16, in <module>
    with Panda(serial=s) as p:
  File "/usr/local/lib/python3.10/dist-packages/pandacan-0.0.10-py3.10.egg/panda/__init__.py", line 257, in __init__
    self.connect(claim)
  File "/usr/local/lib/python3.10/dist-packages/pandacan-0.0.10-py3.10.egg/panda/__init__.py", line 310, in connect
    self._mcu_type = self.get_mcu_type()
  File "/usr/local/lib/python3.10/dist-packages/pandacan-0.0.10-py3.10.egg/panda/__init__.py", line 712, in get_mcu_type
    raise ValueError(f"unknown HW type: {hw_type}")
ValueError: unknown HW type: bytearray(b'\x00')
```

briskspirit, 2023-09-06:
@gcormier you need to flash with DFU, seems that your FW is pretty old and doesn't report HW_TYPE
@ServerDummy are there files in board/obj/ directory? namely panda.bin.signed ?

adeebshihadeh, 2023-11-11:
Paw schematic is in commaai/neo. Check out the community Discord if you need further help.

closed 2023-11-11 by adeebshihadeh