commaai/panda #1652: h7 panda freeze at logging_init if flash was corrupted(?)

briskspirit, 2023-09-06:
After subsequent flashes of the device panda freezes after jump to app.
Narrowed it down to logging module init, this line: https://github.com/commaai/panda/blob/72f1603a89c7fecb6746d0154232f2d8187ebe70/board/drivers/logging.h#L68C8-L68C8
So it freezes before enabling interrupts and USB.
Can be fixed by executing logging_erase() before that loop.
So my suspicion that flash gets somehow corrupted (when writing and panda gets reset command/powered off?)
Wasn't able to reproduce it yet, but got lucky to trip it once.

adeebshihadeh, 2023-09-13:
It's hitting `DBECCERRIE`, a flash ECC error that raises a Bus fault. I'm not sure we can gracefully handle or recover from these ECC errors. 

https://github.com/commaai/panda/pull/1661 will allow pandad to recover from this, but we definitely need to do more here.

adeebshihadeh, 2023-09-15:
The logging is really only useful right now for debugging bootkick issues. Once those are resolved, I propose we remove the logging to flash stuff. We can keep it around in case we need it again.

adeebshihadeh, 2023-10-02:
Removed the flash logging.

closed 2023-10-02 by adeebshihadeh