commaai/panda #1883: Mutation tests on safety unit tests

adeebshihadeh, 2024-02-21:


mpetkovic96, 2024-04-08:
@adeebshihadeh Could you provide more information about this issue in general? Which part of the panda repository should be tested?

maxime-desroches, 2024-10-01:
Currently using these mutations operators (others are not working correctly):
```
mutators: [cxx_increment, cxx_decrement, cxx_comparison, cxx_bitwise_assignment, cxx_bitwise, cxx_arithmetic_assignment, cxx_arithmetic]
```
See this [page](https://mull.readthedocs.io/en/latest/SupportedMutations.html) for a list of all the ops

Here are the current mutations that are not caught by the tests:
[FAILURES_test_body.txt](https://github.com/user-attachments/files/17217274/FAILURES_test_body.txt)
[FAILURES_test_chrysler.txt](https://github.com/user-attachments/files/17217275/FAILURES_test_chrysler.txt)
[FAILURES_test_defaults.txt](https://github.com/user-attachments/files/17217276/FAILURES_test_defaults.txt)
[FAILURES_test_elm327.txt](https://github.com/user-attachments/files/17217277/FAILURES_test_elm327.txt)
[FAILURES_test_ford.txt](https://github.com/user-attachments/files/17217278/FAILURES_test_ford.txt)
[FAILURES_test_gm.txt](https://github.com/user-attachments/files/17217279/FAILURES_test_gm.txt)
[FAILURES_test_honda.txt](https://github.com/user-attachments/files/17217280/FAILURES_test_honda.txt)
[FAILURES_test_hyundai.txt](https://github.com/user-attachments/files/17217281/FAILURES_test_hyundai.txt)
[FAILURES_test_hyundai_canfd.txt](https://github.com/user-attachments/files/17217282/FAILURES_test_hyundai_canfd.txt)
[FAILURES_test_mazda.txt](https://github.com/user-attachments/files/17217283/FAILURES_test_mazda.txt)
[FAILURES_test_nissan.txt](https://github.com/user-attachments/files/17217284/FAILURES_test_nissan.txt)
[FAILURES_test_subaru.txt](https://github.com/user-attachments/files/17217285/FAILURES_test_subaru.txt)
[FAILURES_test_subaru_preglobal.txt](https://github.com/user-attachments/files/17217286/FAILURES_test_subaru_preglobal.txt)
[FAILURES_test_tesla.txt](https://github.com/user-attachments/files/17217287/FAILURES_test_tesla.txt)
[FAILURES_test_toyota.txt](https://github.com/user-attachments/files/17217288/FAILURES_test_toyota.txt)
[FAILURES_test_volkswagen_mqb.txt](https://github.com/user-attachments/files/17217289/FAILURES_test_volkswagen_mqb.txt)
[FAILURES_test_volkswagen_pq.txt](https://github.com/user-attachments/files/17217290/FAILURES_test_volkswagen_pq.txt)

closed 2025-02-20 by sshane