commaai/panda #1956: Panda DFU recovery

dzid26, 2024-05-26:
Panda doesn't always enter DFU on C3X when running `recover.py`.

adeebshihadeh, 2024-08-02:
Are you connected to a PC? Might be a race condition with USB enumeration in the ST bootloader.

[image]

https://www.st.com/resource/en/application_note/an2606-stm32-microcontroller-system-memory-boot-mode-stmicroelectronics.pdf

dzid26, 2024-08-03:
This is on C3X.  I have it connected to 12V for this test.  

Seems to me this is an SPI communication issue:

- I found that `recover.py` gets stuck when `pandad.py` is active.  
- Also, `pandad.py` errors out when that happens.
```
selfdrive/pandad/spi.cc:   1 / 0x0 / 7 / 12 / tx: 14141414141414
selfdrive/pandad/spi.cc: SPI: timed out waiting for ACK, waiting for 0x1f
selfdrive/pandad/spi.cc:   1 / 0x0 / 7 / 12 / tx: 14141414141414
selfdrive/pandad/spi.cc: SPI: timed out waiting for ACK, waiting for 0x1f
selfdrive/pandad/spi.cc:   1 / 0x0 / 7 / 12 / tx: 14141414141414
selfdrive/pandad/spi.cc: SPI: timed out waiting for ACK, waiting for 0x1f
selfdrive/pandad/spi.cc:   1 / 0x0 / 7 / 12 / tx: 14141414141414
selfdrive/pandad/spi.cc: SPI: timed out waiting for ACK, waiting for 0x1f
selfdrive/pandad/spi.cc:   1 / 0x0 / 7 / 12 / tx: 14141414141414
and so on...........
```
- After killing panda.py, the recover.py will show 0 dfu devices.  
- Only after relaunching pandad.py and then killing it first, the recover.py will start to list dfu devices again.
- Resetting panda via nrst pin also fixes "recover.py showing 0 dfu devices"

Looks like some clash when accessing SPI.

adeebshihadeh, 2024-08-03:
Oh yeah, this is expected since the ST bootloader state machine is super sensitive. `recover.py` doesn't take an exclusive lock over the SPI device, so `pandad`'s comms still go through.

closed 2024-08-07 by adeebshihadeh