commaai/panda #2006: getting an error No module named 'request'

hedgehogdrfrink, 2024-08-31:
getting an error when trying to setup a connection with comma 3x to pull files off of it,

./som_debug.sh 
Traceback (most recent call last):
  File "/home/user/Downloads/openpilotv1/panda/tests/./debug_console.py", line 9, in <module>
    from panda import Panda
  File "/home/user/Downloads/openpilotv1/venv/lib/python3.11/site-packages/panda/__init__.py", line 1, in <module>
    from request import PandaRequest
ModuleNotFoundError: No module named 'request'

Followed along with the readme to set it up, not sure what I'm doing wrong at this point.

adeebshihadeh, 2024-08-31:
You have this installed: https://pypi.org/project/panda/. It's not using this panda library.

closed 2024-08-31 by adeebshihadeh

hedgehogdrfrink, 2024-08-31:
Thank you, fixed the issue