commaai/panda #2105: cppcheck: catch ints as bools

adeebshihadeh, 2024-12-13:
Context in #1876

This is three steps:
1. add support for this to cppcheck
2. upstream that support to cppcheck
3. PR to enable it and fix all the violations

While we're waiting for them to merge the support upstream, you can open a draft PR using the patch.

anshgoyalevil, 2024-12-16:
Hey @adeebshihadeh I am willing to resolve this issue. Can you please assign it to me?

anshgoyalevil, 2024-12-16:
I have created a pull request for getting the required changes merged into upstream cppcheck. Lets see how it goes.
https://github.com/danmar/cppcheck/pull/7110

varshneydevansh, 2026-04-25:
The upstream cppcheck fixes are merged, but the latest cppcheck 
release available to package still predate the second fix (`e41115d` 
from cppcheck-opensource/cppcheck#8388) which is `2.20.0`.

I’ll wait for the next cppcheck release that includes `e41115d`, then 
open the dependencies PR and update the panda PR to use it.