commaai/panda #2111: Error: CAN packet version mismatch: panda's firmware v0, library v4. Reflash panda.

blackandwhitehat, 2024-12-16:
Flashed Official White Panda with commit bedee405 and it shows a protocol mismatch. 

`┌──(panda_env)─(panda㉿Depot)-[~]
└─$ python pandafwtest.py 
INFO: connecting to panda 290025000e51363338383037
Panda firmware version: DEV-bedee405-DEBUG

┌──(panda_env)─(panda㉿Depot)-[~]
└─$ python test.py
INFO: connecting to panda 290025000e51363338383037
Panda connected and configured.
Error: CAN packet version mismatch: panda's firmware v0, library v4. Reflash panda.
Panda connection closed.`

Located check in python/__init__.py line 373 

`    # ensure our protocol version matches the panda
    if handle is not None and not ignore_version:
      if spi_version != handle.PROTOCOL_VERSION:
        err = f"panda protocol mismatch: expected {handle.PROTOCOL_VERSION}, got {spi_version}. reflash panda"
        raise PandaProtocolMismatch(err)`

If I bypass this check in __init__.py I get LIBUSB1 intermittent timeout errors and I'm not sure it's actually sending any data to bus.

`┌──(panda_env)─(panda㉿Depot)-[~]
└─$ python test.py
INFO: connecting to panda 290025000e51363338383037
Panda connected and configured.
Message sent successfully.
Panda connection closed.

┌──(panda_env)─(panda㉿Depot)-[~]
└─$ python test.py
INFO: connecting to panda 290025000e51363338383037
Panda connected and configured.
Error: LIBUSB_ERROR_TIMEOUT [-7]
Panda connection closed.`

upsidedownlemon on Comma Discord helped me locate last commit before CAN PACKET VERSION = 1  

[https://github.com/commaai/panda/blob/5f5f30eedc72f62c7a10dec0c57ffeeb7a92faf6/python/__init__.py](url)

Tried to install the older version of pandacan but it failed due to a Libusb1 version error

`┌──(panda_env)─(panda㉿Depot)-[~/panda]
└─$ pip install .
Processing /home/panda/panda
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Collecting libusb1==1.6.6 (from pandacan==0.0.9)
  Using cached libusb1-1.6.6.tar.gz (56 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... error
  error: subprocess-exited-with-error

  × Getting requirements to build wheel did not run successfully.
  │ exit code: 1
  ╰─> [1 lines of output]
      error in libusb1 setup command: use_2to3 is invalid.
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
error: subprocess-exited-with-error

× Getting requirements to build wheel did not run successfully.
│ exit code: 1
╰─> See above for output.

note: This error originates from a subprocess, and is likely not a problem with pip.`

I manually edited the setup.py and requirements.txt to use libusb1-1.5.0. Installed successfully. 

This resolved the issue and panda works as expected now.

analytic-bias, 2025-04-28:
When you say you bricked the white panda, does it look like this? https://github.com/user-attachments/assets/db9992e1-0d6a-4832-96b9-32409b52a3c3

blackandwhitehat, 2026-05-21:
The Discord is where I was told to submit this detailed bug report in the first place. This is a real bug and bricked hardware I paid for right before submitting.