commaai/panda #2200: unable to flash panda after DBC refactor

dkiiv, 2025-05-17:
Hi, since a large section of the safety files has been moved into commaai/opendbc, I am now unable to flash / recover panda from this repository

If there is a new working process, the existing README in board/ is outdated

dkiiv, 2025-05-17:
`board/jungle/main.c:4:10: fatal error: safety.h: No such file or directory
    4 | #include "safety.h"
      |          ^~~~~~~~~~
compilation terminated.
scons: *** [board/jungle/main-panda_jungle.o] Error 1
board/jungle/main.c:4:10: fatal error: safety.h: No such file or directory
    4 | #include "safety.h"
      |          ^~~~~~~~~~
compilation terminated.
scons: *** [board/jungle/main-panda_jungle_h7.o] Error 1
board/main.c:12:10: fatal error: safety.h: No such file or directory
   12 | #include "safety.h"
      |          ^~~~~~~~~~
compilation terminated.
scons: *** [board/main-panda.o] Error 1
board/main.c:12:10: fatal error: safety.h: No such file or directory
   12 | #include "safety.h"
      |          ^~~~~~~~~~
compilation terminated.`

i tried manually adding the safety.h file back, but this didnt work as the refactor ripped more than just that from the repo

`In file included from board/jungle/main.c:4:
board/safety.h:3:10: fatal error: safety_declarations.h: No such file or directory
    3 | #include "safety_declarations.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
scons: *** [board/jungle/main-panda_jungle.o] Error 1
In file included from board/jungle/main.c:4:
board/safety.h:3:10: fatal error: safety_declarations.h: No such file or directory
    3 | #include "safety_declarations.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
scons: *** [board/jungle/main-panda_jungle_h7.o] Error 1
In file included from board/main.c:13:
board/safety.h:3:10: fatal error: safety_declarations.h: No such file or directory
    3 | #include "safety_declarations.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
scons: *** [board/main-panda.o] Error 1
In file included from board/main.c:13:
board/safety.h:3:10: fatal error: safety_declarations.h: No such file or directory
    3 | #include "safety_declarations.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.`

adeebshihadeh, 2025-05-18:
Fixed after https://github.com/commaai/panda/pull/2202

closed 2025-05-18 by adeebshihadeh