commaai/panda #2239: Setup proper pip package

adeebshihadeh, 2025-07-25:
`pandacan` is already a pip package, but it doesn't include the FW and some other things to fully use the panda project.

The package should be sufficient to remove panda as a submodule from openpilot and add it as a normal Python dependency to the pyproject.toml. For validation, make an openpilot PR that does this and ensure CI passes.

shreyanshjain7174, 2025-07-26:
@adeebshihadeh can I work on this one ?

shreyanshjain7174, 2025-07-26:
Created PR #2241 to implement comprehensive pip package with all necessary components including firmware, certificates, and build tools. The package enables openpilot to replace git submodule usage with standard pip installation.

shreyanshjain7174, 2025-07-26:
Updated implementation in PR #2242 with clean, minimal changes:

✅ **Only essential files modified:**
- `pyproject.toml` - comprehensive package configuration
- `python/__init__.py` - optional opendbc dependency support

✅ **Comprehensive pip package features:**
- All firmware binaries, certificates, build tools included
- Optional dependency structure: `pip install pandacan[full]`
- CLI entry points: `panda-flash`, `panda-recover`
- Tested package build and import functionality

This implementation enables openpilot to replace git submodule usage with standard pip installation.

iamrajiv, 2025-08-05:
hey @adeebshihadeh is this still open i can work on this

jaumebalust, 2025-08-20:
I have prepared https://github.com/commaai/openpilot/pull/36034 that can be used when pandacan 0.0.10 is published on pypi

prayanshchh, 2025-09-18:
Hi I am working on the issue, I have built the firmware, right now when python -m build runs, the firmware gets built and we get .elf, .signed and .bin files in the whl, when I test this panda build in openpilot I noticed that we have these imports in

```
openpilot/tools/cabana/panda.h

#include "panda/board/health.h"
#include "panda/board/can.h"
```
do u want .h files in the panda build?

adwit-agg, 2026-04-25:
Pushing  commaai/panda#2392 and commaai/openpilot#37908. All CI checks are green.

Noticed a lot of recent bounty attempts seem to be bots/untested code, so I wanted to throw my hat in with something actually verified. I’ve tested the build locally with the new wheel, full openpilot build is exiting successfully in a clean venv.

It’s a pretty lightweight fix (~10 lines per repo). I used LLMs to help navigate the SCons logic as I don't have any prior experience, but the validation (both code and successful build) is all manual. Fix is pretty straightforward, just ensuring build paths are resolving correctly.

Would love feedback if there is further work needed or if I totally missed something. Else, hope this clears this long overdue bounty!